rosswarren/epevermodbus #1: syntax error on raspberry pi 4

RonGroen, 2022-03-08:
Everything installed: pip install epevermodbus

# python --version
Python 3.7.3
Running gives:
epevermodbus -h
Traceback (most recent call last):
  File "/usr/local/bin/epevermodbus", line 6, in <module>
    from epevermodbus.command_line import main
  File "/usr/local/lib/python2.7/dist-packages/epevermodbus/command_line.py", line 18
    print(f"Solar voltage: {controller.get_solar_voltage()}V")
                                                            ^
SyntaxError: invalid syntax
On raspberry Phyton 2 and 3 are installed. How do I install epevermodbus on python3?

RonGroen, 2022-03-08:
solved
install with:
 pip3 installepevermodbus

closed 2022-03-08 by RonGroen

rosswarren, 2022-03-08:
Thank you @RonGroen I will update the README to clarify this a bit more