rosswarren/epevermodbus #8: Testing local changes

rosswarren, 2022-08-28:
There should be a way to test local changes without having to publish to PyPI or making temporary changes to the code.
Currently, the import defaults to the installed version of the library.

hrford, 2022-08-28:
I've been testing my example module by running a python venv from the directory above itself:

`/(some_other_dirs)/epevermodbus$ ../bin/python examples/write_battery_params.py`
Or
`/(some_other_dirs)/epevermodbus$ ../bin/python .py epevermodbus/command_line.py`

The import will then operate on the subdirectory epevermodbus:
`from epevermodbus.driver import EpeverChargeController`

hrford, 2022-08-28:
Turns out that doesn't really cut it... Still needs work :/

SmithChart, 2023-10-24:
When I am working on python packages I usually have them installed editable in a venv:

```
$ git clone https://github.com/rosswarren/epevermodbus.git
$ cd epevermodbus
$ python3 -m venv venv
$ source venv/bin/activate
$ pip install -e .
$ pip list
Package           Version Location
----------------- ------- -----------------------------------
(...)
epevermodbus      0.0.16  /path/to/epevermodbus
```

On every restart of your test the new code from epevermodbus will be used.
Does this help?