rosswarren/epevermodbus #15: Req: Write Coils for Load On/Off Switching

na7q, 2023-03-11:
A load switch would be a great addition to this project.

sante85, 2023-03-17:
@na7q  you want active this in attachment?
[image]

GitHubUser655321, 2024-01-18:
I'm here for the same reason, to control the Load Control Relay.

The documentation says the address is 2;
```
Number: H3
Variable Name: Manual control the load
Address: 2
Description: When the load is manual mode，1-manual on 0 -manual off
```
To turn it on I, I try;
`self.write_registers(0x2, [1])`

I get error;
`minimalmodbus.IllegalRequestError: Slave reported illegal data address`

I've tried;
`self.write_registers(0x02, [1])`
with the same results.

rosswarren, 2024-01-19:
Hey @GitHubUser655321 write_registers uses function code 16 and writes integer values. The Control load should be using function code 5. You could try with `self.write_bit` rather than `self.write_registers` and pass 0 or 1.

GitHubUser655321, 2024-01-20:
Hey @rosswarren - thanks, that worked great!

I put this code in my driver.py file and it does the job;

```
    def set_manual_control_the_load(self, turn_on):
        if turn_on:
            return self.write_bit(0x2, 1)
        else:
            return self.write_bit(0x2, 0)
```

I'm sure you can make it more elegant if you decide to add it.

Thanks again!