rosswarren/epevermodbus #20: Read / Set controller clock

jnhmn, 2023-08-26:
I've implemented functions to read/set the clock of the charge controller to get valid results for the daily/weekly statistics.

```
  def get_clock(self):
  
    def split_short(val):
      hext = '{0:04x}'.format(val)
      ha = hext[:-2]
      hb = hext[-2:]
      return (int(ha,16), int(hb,16))
  
    r = self.retriable_read_registers(0x9013,3,3)
    min, sec = split_short(r[0])
    day, hour = split_short(r[1])
    year, mon = split_short(r[2])
    year += 2000

    curDate = datetime(year,mon,day,hour,min,sec)
    return curDate
```

```
 def set_clock(self, time):
  
    def gen_hex(a, b):
      ha = '{0:02x}'.format(a)
      hb = '{0:02x}'.format(b)
      return ha + hb
  
    r3 = int(gen_hex(time.year-2000,time.month), 16)
    r2 = int(gen_hex(time.day,time.hour), 16)
    r1 = int(gen_hex(time.minute,time.second), 16)
    self.write_registers(0x9013,[r1, r2, r3])
```

Can you please include these 2 functions.

I did not create a pull request since I'm not sure if this meets your requirements for clean code.

na7q, 2023-08-29:
This is great! I'll have to give it a try when I get a chance. If good, I'll merge in into mine version. Thanks for doing this!

brynnneilson, 2023-10-22:
I'm after this functionality. Can you let me know how to add your patch?

SmithChart, 2023-11-03:
I've added this in #21.
you can use my next-branch that combines my PRs and #16 from here: https://github.com/SmithChart/epevermodbus/tree/next
Try to install it with: `pip install git+https://github.com/SmithChart/epevermodbus.git@next`

PR #21 (2023-11-03): Add support to read and write more values via library and command line
This patch adds support to read and write the RTC. Documentation has been taken from:
https://www.img4.cz/i4wifi/attach/StoItem/7124/MODBUS-Protocol-v25.pdf

This change has been tested on my EPEVER Tracer2210AN.

This will also close #20 .
closed 2023-11-03 by rosswarren