rosswarren/epevermodbus #25: Change EPEVER slave address

Janl1, 2024-03-31:
Hello,

is there a way to change the slave address with this tool?

asscherhofstede, 2024-04-08:
via the command line with argument --slaveaddress 

`usage: epevermodbus [-h] [--portname PORTNAME] [--slaveaddress SLAVEADDRESS]

optional arguments:
  -h, --help            show this help message and exit
  --portname PORTNAME   Port name for example /dev/ttyUSB0
  --slaveaddress SLAVEADDRESS
                        Slave address 1-247
  --baudrate BAUDRATE   Baudrate to communicate with controller (default is 115200)`

and within the code

`controller = EpeverChargeController(args.portname, {slave address 1-275}, args.baudrate)
`

SmithChart, 2024-04-08:
Or do you mean set a new permanent slave address on the device itself?

One manual (https://www.img4.cz/i4wifi/attach/StoItem/7245/MODBUS-Protocol-v25.pdf) I found says:
```
2. The default controller ID number is "1", we can modify the ID via PC
common software or MT50 LCD unit (Tips: when modify ID, please make sure
the bus only connects a controller. After modifying the ID, please recharge the
controller.)
```
But I does not mention how it is done.
I tried to find out if the MODBUS specification states any standard way. But wasn't successful :-/

Janl1, 2024-04-15:
In my case I wanted to change the slave address of an Epever XTRA3210N Solar Charge Controller.

With some reverse engineering of the official Epever software I found out that you cannot change the slave address to a normal register value, but you can use the function code 69 to change it.

I used the following line of code to change the address. It will probably return an error message because it cannot parse the result, but the change went through in my case.

instrument._perform_command(69, instrument._num_to_onebyte_string(newId)) (minimalmodbus v2.0.1)

closed 2024-04-15 by Janl1

SmithChart, 2024-04-20:
FTR: I've tried this on my Tracer 2210 and was *not* successful:
```
d = epevermodbus.EpeverChargeController(portname="/dev/ttyUSB0", slaveaddress=1)
d.get_solar_voltage()  # works
d._perform_command(69, b'\x01')
Traceback (most recent call last): (...)
minimalmodbus.InvalidResponseError: Wrong return slave address: 248 instead of 1. The response is: b'\xf8E;\x83r'

e = epevermodbus.EpeverChargeController(portname="/dev/ttyUSB0", slaveaddress=248)  # trying the address from the exception
e.get_solar_voltage()
Traceback (most recent call last): (...)
minimalmodbus.InvalidResponseError: Wrong return slave address: 59 instead of 248. The response is: b';\x04\x02\x04\xb4c\x82'

f = epevermodbus.EpeverChargeController(portname="/dev/ttyUSB0", slaveaddress=59) # again, trying the address from the exception.
f.get_solar_voltage() # but works this time.
12.04
```

The actual ID my device assumes seems to be wildly random. But the EPEVER vendor-software is still able to discover the device and to set a new ID.