rosswarren/epevermodbus #31: Settings Conflict with Epever

jimlynnjulian, 2024-10-30:
I have been working with. So far a solar station that uses Epever MMPT charge controllers.
This year I upgraded froma Tracer 3210N to a Triron 4210N. 
. 
I use multiple scripts in python, php, html and bash to run and monitor the system.
The Triron was treated as a drop-in for the Tracer wiithout any problems. Today, I discovered what may be a difference between the two.
A page that graphs the data had some peculiarities, so I began a search. What I discoverd is that your script ro read settings conflicts with the settings reported by the Epever PC software in two parameters: boost duration and equalization duration, otherwise they are the same.

I suspect, but am not sure, the Triron may use different registers for those two parameters.
Is there a test I can perform?

Also, I think the efforts to write to those two registers fail for the same reason.

SmithChart, 2024-11-05:
Have you tried to hunt down a MODBUS register definition for the Triron?
For the Tracer I have found the following documents:

* https://www.img4.cz/i4wifi/attach/StoItem/7124/MODBUS-Protocol-v25.pdf
* https://www.developpez.net/forums/attachments/p196506d1451307310/systemes/autres-systemes/automation/probleme-com-modbus-pl7-pro/controllerprotocolv2.3.pdf/

With the actual definition at hand we could skip the reverse engineering.

Otherwise you could sniff and analyze the MODBUS communication between the vendor software and the devices and spot differences. We could add a command-line switch or an auto-detection to do the right things for each device.

jimlynnjulian, 2024-11-05:
Thanks for responding.

I'm thinking Wireshark would be a good tool to capture the ethernet communications between my laptop and the Triron.

First, I'm going to communicate directly with Epever and see if they will/can help. Googling the web is the next step.

One thing, that might be called a brute force approach, would be to read all the registers looking for the durations that are set. Perhaps write unusual values to the durations parameters to make them easier to spot in the read. Write something like 333 and 444.

I found this on the web:
"We love Modbus, but its one major hickup is the way it does addressing.  Internally, in the actually bytes that get sent to the device, the addressing is dead simple.  Each of the four types, coil, input status, input register and holding register all have 65535 different 16 bit tag locations available.  But for some reason, when they created Modbus they decided they should create a separate addressing scheme on top of this.  This addressing is solely a documentation thing and has nothing to do with what is sent to the device.  They decided that really, no device is ever really going to need more than 9999 addresses within each type, so lets put a number in front so its obvious which type it is.  So, they put a 4 in front for holding registers, 3 for input registers, 1 for input status and left coils alone.  Why they picked 4 and 3 I don't know since the function for holding register is actually 3, and for input register is 4, which is opposite.  Now if that wasn't enough, for some reason they thought that electricians, who when Modbus was invented back in the 70's were the ones doing all the installs, couldn't count from 0 like programmers.  Why I don't know.  All the electricians I've ever met were very smart people and probably could do math in base 6 if they had to.  Anyhow, since they thought they couldn't count from 0, the offset all the addresses by 1."

https://support.azeotech.com/topic/5845-5-digit-versus-6-digit-modbus-addresses-400001-40001-or-just-1/

SmithChart, 2024-11-05:
Does your Triton have actual Ethernet? The pictures I found on the internet look like RS485 - but on an RJ45-jack (like Ethernet uses).
As far as I know: Wireshark has a dissector for Modbus over TCP/IP (Modbus-TCP), but lacks support for capturing Modbus on RS485 (Modbus-RTU).

Yeah, the addressing scheme is a dumpster fire. To make things worse: Some software wants you to add the leading number, some don't... It's a bit like the 7-bit addresses of I2C. You never know if it's bit 0..7 or 1..8 :-D

jimlynnjulian, 2024-11-05:
Hello,
   The Triron has the standard Epever charge controller connector. I run the Epever PC software by connecting to the Triron using a USB to RJ-45/RS485 cable. My mistake.

jimlynnjulian, 2024-11-05:
Thought I'd mention, the reason I used your software is to replace the somewhat troublesome Epever software. Most of the wrok I do is related to a solar power station I've built. Most of the time, I SSH into a Raspberry Pi 4 and make necessary modifications or maintenance over WiFi. No need to connect the PC with a cable. The software services that run, use the RPi USB cable to the Triron RS485/RJ45 port. I've been doing this for over ten years. Thought the use of SSH would be a useful improvement. I finished rebuilding/configuring the solar array framework today. I'm taking a shhort break before I tackle the next problem. My table is, so to speak, full.
Thanks for your time.

SmithChart, 2024-11-07:
That sounds like a good scheme for remote work on such solar arrays.
This project is open source, so it benefits a lot from people with experience contribution back to it. Looking forward to see your results once you come around to it!

jimlynnjulian, 2024-11-07:
Thanks. I've reorganized my thinking and re-examined the situaion. I use two software libraries with the Triron:
epevermodbus and PHPTracer by Soltoggio at github/toggio.
toggio's does not read durations correctly but epevermodbus does. epevermosbus reads all but does not write any.

If and when I can get back to the issue, I'll delve into modbus more deeply, but just so you(collective form) know, Epever sent me the same register numbers as are used in the Tracer, 906B and 906C.

Anyway, thanks for responding.
Enjoy the holidays.

jimlynnjulian, 2024-11-08:
[image]

closed 2025-07-16 by jimlynnjulian